SlicerIGT/LumpNav #7: Markups should not be visible on slice view

lassoan, 2015-08-31:
Red circles (the added markups) appear in the slice view in contouring mode. As it's quite random when the markups are exactly in the slice plane, they randomly appear/disappear, which is not nice. Turn off visibility of the markups to avoid this.

brudfors, 2015-09-01:
Hello Andras,

Regarding these issues. Will you start working on them at PerkLab? If not, I could direct these tasks to the two new people here in my lab (who will be involved with this project for the foreseeable future). I they would be a good way of getting to know the module.

Cheers,

Mikael

kanyala, 2015-09-01:
Hi MikaelI work on the issues, though more people could do more. I will call a meeting to discuss a bit coordination of activities and what are the needs the module should implement.

—
LK

brudfors, 2015-09-01:
Okay, that sounds great. Please keep me updated.

lassoan, 2015-09-01:
Before you start working on an issue assign it to yourself to make sure that nobody else working on it at the same time.

kanyala, 2015-09-01:
Is it ok if markups visibility is turned off on all panels: navigation, contouring, calibration?
Currently it is turned on on contouring panel, turned off on navigation panel button press and the calibration panel switch did't change the visibility state so it always used the latest visibility status.

Laci

kanyala, 2015-09-02:
fixed in
commit c81d76c3a029360a6a2a9259c547dbdb6c8617a3

closed 2015-09-02 by kanyala

lassoan, 2015-09-02:
Yes, always hide the markups.