SlicerIGT/LumpNav #40: Implement Undo cautery calibration

ungi, 2021-07-30:
The previous CauteryTipToCautery should be saved when a new pivot calibration is started. This button would change the current one to the saved last one.

JoshEhrlich, 2022-03-20:
Where is the best location to save this? Should it be saved in the LumpNav folder, or somewhere personalized to the user outside of the LumpNav directory? 

Also, I am assuming we do not want to keep the saved calibration file over time. It will just be used during a single operation.

ungi, 2022-03-20:
> Where is the best location to save this? Should it be saved in the LumpNav folder, or somewhere personalized to the user outside of the LumpNav directory?
> 

If you reference the old calibration transform in the parameter node, then it will be automatically saved and loaded with the scene. Most commonly, we just need the old calibration for a few minutes until we see the results of the new calibration. Once we see a new calibration, we can immediately decide if we want to use that or go back to the previous one.