SlicerIGT/LumpNav #56: Save scene if there were changes

ungi, 2021-10-29:
Add a parameter to the parameter node that tells if it has been changed since last save or not. If user wants to exit with unsaved changes, then save automatically.

JoshEhrlich, 2021-11-17:
I am a little confused on this. Here is my pseudo code:
-Add observer that watches parameter node (which event?)
-Call function when changed that sets a boolean variable called unsavedData to True/False when the parameter node is changed and the scene is not saved. 
- within the "exit" function, add a check of this variable. Save automatically if not being done by user

My only worry is that won't this start happening even when we are developing? Maybe we could go over some use-cases together.

lassoan, 2021-11-17:
You don't need to add an observer to notice changes. Whenever you modify a node, its modified timestamp is updated automatically. If the node's modified timestamp > last save's timestamp then you need to save it. To save the timestamp of the last save, you can create a timestamp object and whenever you saved, call `Modified()`.

```python
# Have a member variable that stores the last save time
saveTime = vtk.vtkTimeStamp();

# Call this when the data set is loaded and each time you save the data set
saveTime.Modified()

# Check if save is needed:
needToSave = (parameterNode.GetMTime() > saveTime.GetMTime())
```

JoshEhrlich, 2021-11-17:
Wonderful, thank you very much!

commit f4482845 (2022-10-28): Fixed #56: use modified timestamp instead of observer
closed 2022-10-28 by chriscyyeung